andr36oid/releases #1: can test anything

EatPrilosec, 2025-07-13:
i have a few R36S, one specifically i use for UART. i'd love to help out. 

release mentions UART logs, you still need some? ill try to get them and post tonight just in case you do.

I'm wrapping up the first beta of [my multiboot project](https://github.com/R36S-Stuff/R36S-Multiboot) but ill be taking a break on that once the beta is released and would love to help

EatPrilosec, 2025-07-13:
```
U-Boot 2017.09-00048-ge8824042a0-dirty (Oct 07 2021 - 18:03:21 -0400)

Model: Rockchip RK3326 RG351 Series
DRAM:  992 MiB
Sysmem: init
Relocation Offset is: 3dabc000
Using default environment

adc0 (hw rev) 165
Model = rg351mp
RKPARM: Invalid parameter part table
reading rg351mp-kernel.dtb
74204 bytes read in 6 ms (11.8 MiB/s)
Hotkey: ctrl+c
I2c speed: 400000Hz
PMIC:  RK8170 (on=0x10, off=0x04)
vdd_logic 1100000 uV
vdd_arm 1100000 uV
MMC: no card present
mmc_init: -123, time 0
*** Warning - read failed, using default environment

In:    serial
Out:   serial
Err:   serial
Model: Game Console R35S/R36S
dwmmc@ff370000: 1, dwmmc@ff380000: 2
## Error: "rkimg_bootdev" not defined
Bootdev: mmc 0
rockchip_get_bootdev: Can't find dev_desc!
[Vendor ERROR]:Invalid boot device type(0)
rockchip_get_bootdev: Can't find dev_desc!
[Vendor ERROR]:Invalid boot device type(0)
download key pressed... entering recovery mode!
rockchip_get_bootdev: Can't find dev_desc!
rockchip_get_boot_mode: dev_desc is NULL!
CLK: (sync kernel. arm: enter 600000 KHz, init 600000 KHz, kernel 600000 KHz)
  apll 600000 KHz
  dpll 664000 KHz
  cpll 24000 KHz
  npll 1188000 KHz
  gpll 1200000 KHz
  aclk_bus 200000 KHz
  hclk_bus 150000 KHz
  pclk_bus 100000 KHz
  aclk_peri 200000 KHz
  hclk_peri 150000 KHz
  pclk_pmu 100000 KHz
** Can't read partition table on 1:0 **
** Invalid partition 1 **
Rockchip UBOOT DRM driver version: v1.0.1
Using display timing dts
Detailed mode clock 29000 kHz, flags[8000000a]
    H: 0640 0686 0688 0732
    V: 0480 0645 0647 0661
bus_format: 100e
final DSI-Link bandwidth: 192 Mbps x 4
** Can't read partition table on 1:0 **
** Invalid partition 1 **
logo.bmp file not found! filesize = 0

spinor read: device 1 block # 12592, count 400 ... sfc nor id: ff ff ff
ret -1
-1 blocks read: ERROR
cat not find bmp file
show_bmp : failed to decode bmp at 0x000000003df2c000
```

so, it cant read the partition table of the MMC, but it IS reading `rg351mp-kernel.dtb` the error with the partition table comes AFTERwards. from my experience messing with DTB and u-boot for multibooting, that tells me the `rg351mp-kernel.dtb` file is wrong. i noticed its the same size as android, sha summed, and its the same DTB

the uboot dtb for this device HAS TO BE the one that comes on the original sdcard. theres a few devs working on this. the MMCcard address in the uboot devicetree is different for os than it is for u-boot for some reason. i believe rocknix is the only one to have conquered that demon, and i havent asked them how yet lol. 

[heres a link to a known good "u-boot dtb." ](https://github.com/R36S-Stuff/R36S-U-Boot-PanelSupport/blob/main/ScreenFiles/Panel%204/rg351mp-kernel.dtb) (supports "new panel" 4)

i also noticed you didnt include a logo.bmp, iirc, that failure can cause irratic behavior. 
[rocknix, uos, amberelec all patch the show_logo part of the init out of the init process ](https://github.com/ROCKNIX/hardkernel-uboot/blob/611716febddb824a7203d0d3b5d399608a54ccf6/board/rockchip/odroidgoa/odroidgoa.c#L208)

the android logo would be a nice touch, JS.

also, if youre going to make this r36S specific, [panels are a thing.](https://github.com/R36S-Stuff/R36S-U-Boot-PanelSuppor) its a mess.

also, i noticed you're not loading the initrd in u-boot, is that because android doesnt have one? if you're not using it because you noticed the kernel only boots when you don't load it, the reason is there is only 13 mb of space for the initrd using the derault load addresses, and youre overwriting the kernel when it gets bigger. my strategy to overcome this in armbian was to [load it after the end of the kernel](https://github.com/R36S-Stuff/R36S-Armbian/blob/646415bf586b0246599361dbd1790a3f2862f033/userpatches/rk3326/R36S/boot.ini#L23)

sorry, don't know how much you know about this devices u-boot process, so I'm just being thorough, and the prospect of android has me excited, so I'm sharing what I've learned, but I'm very confident i can help you at least get to the android kernel, you'll have to take it from there lol, android gives me a headache. feel free to ask questions, or tell me to test something for you.

sonic011gamer, 2025-07-18:
I already have someone testing this for me, and it was just a test build.

Get excited.

closed 2025-07-18 by sonic011gamer