andr36oid/releases #3: R35S Compatibility

nuvim, 2025-07-23:


sonic011gamer, 2025-07-23:
You most likely have a different panel than the R36s we use to test which has panel 4.
Unfortunately I don't own a R35S so I have no clue how ArkOS handles it.

nuvim, 2025-07-23:
> You most likely have a different panel than the R36s we use to test which has panel 4. Unfortunately I don't own a R35S so I have no clue how ArkOS handles it.

I would like to help but I don't know where to start, can you guide me?

nuvim, 2025-07-24:
I admit I tried doing this using the DTB that came with the R35S, but this happened (no, the R35S didn't get bricked — it just shows the LineageOS logo, which slowly fades out until the screen goes black, and then the console reboots).

[image]

[image]

aksenav2, 2025-07-24:
It doesn't work for me either on the original R36S You have panels 0.1.2 sorted into folders there.....As a result, I went to the site to check what panel I have in digital designation, and I got this window without numbers. So I had to try all the panels at random. Surprisingly, 0 and 5 panels suited me, but on the fifth, when turning on the android, there is even a small short vibration, like on many smartphones, so I assume - it is it, but on none, all the same, android does not fully start, the first launch and after a while a reboot, and on the second launch - a logo and then a long dark screen and eventually, after 2-3 minutes the console turns off.

[image]

nuvim, 2025-07-24:
> It doesn't work for me either on the original R36S You have panels 0.1.2 sorted into folders there.....As a result, I went to the site to check what panel I have in digital designation, and I got [...]

the LineageOS logo fades out and then the console reboots after a few minutes.
I also tried using the DTB that came with my R35S, but no success. I’m planning to explore how the DTB files work to see if there’s a way to make it compatible with our model.

closed 2025-07-25 by nuvim

nuvim, 2025-07-25:
the new version is working on r35s (I USED PANEL 5)

aksenav2, 2025-07-25:
On the forum, they sent me the panel file 0 for the original and now the logo has appeared and you can see that the startup is in progress, but when the desktop is displayed, the console simply turns off. And sometimes it turns off even before the desktop appears. 

[panel_0.zip](https://github.com/user-attachments/files/21423410/panel_0.zip)

nuvim, 2025-07-26:
> On the forum, they sent me the panel file 0 for the original and now the logo has appeared and you can see that the startup is in progress, but when the desktop is displayed, the console simply [...]

you need to try other panel, i tried all of this panel and the panel 5 works for me

aksenav2, 2025-07-26:
> you need to try other panel, i tried all of this panel and the panel 5 works for me

Thanks for the advice, but I have already found a solution. First, I run the firmware on any kernel, but in order to fully start the console to the initial settings, even though there is a dark screen on the display, but the system boots, so I press the power button and then press the confirmation button. After that, you can throw in the kernel and panel for your revision of the console. Everything is simple and ingenious, but it took a lot of tests to come to this.