andr36oid/releases #17: I can't use my usb wifi

mrbitnasa-a11y, 2025-08-03:
I just wanna test onl game. But I cant use wifi when I plug it in nothing happen pls help

mrbitnasa-a11y, 2025-08-08:
[image]

sonic011gamer, 2025-08-19:
Next release I'll have it most likely fixed.

ehcerh57jdjshG, 2025-08-19:
Is it possible to add support for the TP-link tl-wn823n? 

[image]

sonic011gamer, 2025-08-19:
> Is it possible to add support for the TP-link tl-wn823n? 
> 
> 

Do as the people above, install deviceinfohw and send me the id of the usb

ehcerh57jdjshG, 2025-08-19:
oh oops sorry my bad 🫣

[image]

thank you in advance!