andr36oid/releases #26: Panel X doesn't turn on.

AwayGloom, 2025-08-25:
**What panel are you u

[rk3326-r35s-linux.zip](https://github.com/user-attachments/files/21962009/rk3326-r35s-linux.zip)

ing?**
Tell us what panel you're using.
- [ ] Panel 0
- [ ] Panel 1
- [ ] Panel 2
- [ ] Panel 3
- [x] Panel 5

## Have you tried replacing your DTBs?
- [x] I have tried replacing other panels from the BOOT partition.

# Attach the DTBs you use in ArkOS, otherwise I will close the issue.

AwayGloom, 2025-08-25:
I got some DTBS that made the R36S available. The logo appears, but a red light flashes and the screen goes off, but the backlight is very dim.
I went into the files and it's creating the normal Android.
Elida panel, kd35t133\0simple-panel-dsi

sonic011gamer, 2025-08-31:
Try the new pre-release build.