andr36oid/releases #30: Panel X doesn't turn on.

DES362, 2025-09-03:
**What panel are you using?**
Tell us what panel you're using.
- [ ] Panel 0
- [ ] Panel 1
- [ ] Panel 2
- [ ] Panel 3
- [x] Panel 4
- [ ] Panel 5

## Have you tried replacing your DTBs?
- [x

[rk3326-r36s-android.zip](https://github.com/user-attachments/files/22109733/rk3326-r36s-android.zip)

] I have tried replacing other panels from the BOOT partition.

# Attach the DTBs you use in ArkOS, otherwise I will close the issue.

--- lil extra info
the only ones that actually turned on the screen in a non static-ish way was panel 4 and the plus. they turned it on, it started to load, and then it either crashed or turned off

DES362, 2025-09-03:
sorry i messed up the format a bit, the file seemed to have uploaded at the "Have you tried replacing your DTBs?" section. for clarification i did try the other DTBs

sonic011gamer, 2025-09-03:
Check your sd card, does ezroms exist?
If no, then try wiping your sd clean, and format your drive as gpt

DES362, 2025-09-03:
> Check your sd card, does ezroms exist? If no, then try wiping your sd clean, and format your drive as gpt

yes my sd card does have ezroms, however it has no folders, nor files inside

sonic011gamer, 2025-09-05:
Try the old release, if it works then you need to clean the drive first and convert it to gpt before flashing

DES362, 2025-09-05:
do u have any other ways to fix it? it may be my r36s's fault, but i bought a legit one so im not sure